CyanideCN/PyCINRAD #6: 沈阳雷达数据格式

CyanideCN, 2018-09-04:
[image]
在中国气象数据网上看到沈阳雷达是SC雷达，但是用THREDDS读取时提示为SA雷达。
[image]
但是这个雷达的参数并不符合SA雷达的参数，在以SA雷达读取时没有报错但是读出来的数据完全不对。
（此雷达库长250m，库数1000个）

CyanideCN, 2018-09-04:
这个基数据在101-109字节并不是CINRAD/SC或CINRAD/CD，所以排除SCD类型。同时，在117-125字节不是CINRAD/CC，排除CD类型。该文件长度为10147680字节，并不能被SAB雷达单个数据块的2432字节整除，也不能被CAB雷达单个数据块4132字节整除，所以也不是SAB或者CAB类型。现在完全不知道这个基数据是什么类型的。

zhaopingsun, 2018-11-30:
沈阳是784厂的，型号是SC

1782005668, 2019-04-24:
请问，您上边贴出来的数据图，是用什么软件打开的啊

lovechang1986, 2019-09-28:
Mi4是可以读取的，据说是3132字节和ws88d的2432很像，但格式还是有所差别。用mi4读取9个仰角，360根Ray，同时算了一下距离库大概160km左右，只能这么推断，我在气象家园问了，但还没回复。我觉得最终解决只能是问问开发mi4的人，找找这个格式。

CyanideCN, 2019-09-28:
@lovechang1986 刚刚试了一下，文件头应该是SAB系列的
[image]

lovechang1986, 2019-09-28:
:+1:，这是16字节的文件头？你看我在气象家园的悬赏，我安装3132字节全部都是数据，可视化后中间有一部分就是很大的，这部分应该就是文件头，还有2圈很像的我也不知道是啥了。

lovechang1986, 2019-09-28:
只是不知道为什么还要2圈重复的。如果搞明白这个，这个格式就算是强行破解了:joy:

CyanideCN, 2019-09-28:
@lovechang1986 我按照气象家园那人说的规则试了一下，和SAB文件规则一样，应该算是解决了，我整理一下代码然后更新:laughing:
[image]

lovechang1986, 2019-09-28:
强大，现在电脑没有互联网，等你更新

CyanideCN, 2019-09-28:
@lovechang1986 新的模块whl可以在[appveyor](https://ci.appveyor.com/project/CyanideCN/pycinrad/builds/27731444/job/856uwc5p66abc9a1/artifacts)下载，从github拉取自己编译也行

PR #43 (2019-09-28): Decode special SC files
Close #6 . Close #40
closed 2019-09-28 by CyanideCN