CyanideCN/PyCINRAD #7: Add mapping function

zxdawn, 2018-09-06:
Your repository supports many types of radars now, but doesn't have the mapping function like [this](https://github.com/zxdawn/pyart/blob/CNRAD/notebooks/Using_crfmapdisplay_to_map_many_S-band_radars_to_a_Cartesian_grid.ipynb). You can find the original link of pyart [here](http://arm-doe.github.io/pyart/dev/auto_examples/mapping/plot_map_two_radars_to_grid.html#example-mapping-plot-map-two-radars-to-grid-py).

CyanideCN, 2018-09-06:
Thanks for your appreciation. Actually it's a quite good idea, so I may consider to support this function. From existing function which calculates composite reflectivity, a little adjustment that remaps the data to regularly-spaced latlon grid instead of equidistant grid is enough.

CyanideCN, 2018-09-07:
The problem now I am facing is that using 'nearest' method in scipy.interpolate.griddata will interpolate regions outside the radar observational range. For single radar, the problem is easy to solve by masking the grids that are more than 230 km away from radar. But talking about radar mosaic, this is not easy to archieve because multiple center points exist. I'm wondering if you have some ideas to mask grids that are out of range in this case.

zxdawn, 2018-09-07:
@CyanideCN You can check the parameter `grid_limits` in [pyart](https://github.com/ARM-DOE/pyart/blob/29869a5a9dd9076d37d1a425058c4e14c75b87a6/pyart/map/grid_mapper.py#L57)

CyanideCN, 2018-09-07:
Actually I'm quite confused about how it works... I'm not familiar to pyart, so could you help me about this part? If you are interested, I can add you as the collaborator of this respository.
PS: I just come out with an idea that masks outside grids by creating a masked array using the calculated geographic coordinate points which lie in the range. About the weighting function, I really have no idea about how to implement this.

zxdawn, 2018-09-07:
@CyanideCN OK, You can add me as the collaborator. But, I'm working on some projects now and this function will be delayed if you aren't in a hurry.
Actually, your idea should work.

PR #36 (2019-06-24): Map radars into grid
Close #7
closed 2019-06-24 by CyanideCN