CyanideCN/PyCINRAD #17: 自定义图片保存路径

CyanideCN, 2018-09-25:
延续 #16 的讨论。
在注释掉`save`函数之后，`plt.savefig`所得到的图片会出现文字错位的现象。

CyanideCN, 2018-09-26:
@shevawen 找到一个问题，在每次运行结束之后设置好的环境变量会消失，在下一次使用的时候又会切换到用户目录。
```python
import os
os.environ['CINRAD_PATH']
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\27455\AppData\Local\Programs\Python\Python36\lib\os.py", line 669, in __getitem__
    raise KeyError(key) from None
KeyError: 'CINRAD_PATH'
os.getenv('CINRAD_PATH')
import cinrad
os.environ['CINRAD_PATH']
'C:\\Users\\27455\\PyCINRAD\\'
```

CyanideCN, 2018-09-27:
@shevawen 之前坐标轴错误的问题可能是升级到matplotlib 3.0导致的，matplotlib 2.2保存下来的图片坐标轴的位置是对的，但是图片的右边一部分丢失了。
[image]

shevawen, 2018-10-08:
我复现的效果是这样的。本周继续跟进哈~
[image]

CyanideCN, 2018-10-08:
@shevawen 你这个效果还正常一点，貌似和matplotlib的版本也有一定关系。从你这张图来看，好像是`text`函数执行在`colorbar`之前了，导致标注的文字是按照`geoaxes`的坐标轴来定位而不是`colorbar`的坐标轴。

commit 5b59961f (2018-10-15): Close #17

使用PPI和RHI类的__call__方法来自定义图片保存路径
closed 2018-10-15 by CyanideCN