CyanideCN/PyCINRAD #35: 使用PPI绘制PUP基反产品出错

laishenggx, 2019-05-15:
`from cinrad.io import PUP
from cinrad.visualize.ppi import PPI

f=PUP('C:\\pyproj\\cinrad\\pupdata\\R\\19\\20170330.084822.01.19.778')
data=f.get_data()
print(data.data.shape)
fig = PPI(data)
fig('C:\\pyproj\\cinrad\\')`

此时解释器提示
C:\Users\win\Anaconda3\python.exe C:/pyproj/cinrad/pup_ppi.py
C:\Users\win\Anaconda3\lib\site-packages\dask\config.py:168: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  data = yaml.load(f.read()) or {}
C:\Users\win\Anaconda3\lib\site-packages\distributed\config.py:20: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  defaults = yaml.load(f)
(366, 230)
Traceback (most recent call last):
  File "C:/pyproj/cinrad/pup_ppi.py", line 8, in <module>
    fig = PPI(data)
  File "C:\Users\win\Anaconda3\lib\site-packages\cinrad\visualize\ppi.py", line 94, in __init__
    self._plot(**kwargs)
  File "C:\Users\win\Anaconda3\lib\site-packages\cinrad\visualize\ppi.py", line 148, in _plot
    self.geoax.pcolormesh(lon, lat, var, norm=pnorm, cmap=pcmap, **kwargs)
  File "C:\Users\win\Anaconda3\lib\site-packages\cartopy\mpl\geoaxes.py", line 1459, in pcolormesh
    result = self._pcolormesh_patched(*args, **kwargs)
  File "C:\Users\win\Anaconda3\lib\site-packages\cartopy\mpl\geoaxes.py", line 1491, in _pcolormesh_patched
    X, Y, C = self._pcolorargs('pcolormesh', *args, allmatch=allmatch)
  File "C:\Users\win\Anaconda3\lib\site-packages\matplotlib\axes\_axes.py", line 5575, in _pcolorargs
    C.shape, Nx, Ny, funcname))
TypeError: Dimensions of C (366, 230) are incompatible with X (366) and/or Y (230); see help(pcolormesh)

Process finished with exit code 1

CyanideCN, 2019-05-16:
我觉得可能是生成经纬度数据的时候出了问题，方位角和距离是手动生成的，也许形状出错了。
https://github.com/CyanideCN/PyCINRAD/blob/master/cinrad/io/io.py#L669

laishenggx, 2019-05-16:
> 我觉得可能是生成经纬度数据的时候出了问题，方位角和距离是手动生成的，也许形状出错了。
> https://github.com/CyanideCN/PyCINRAD/blob/master/cinrad/io/io.py#L669

请问有没有示例的脚本呀？或者说是用法说明的

CyanideCN, 2019-05-16:
用法在README里面有，PUP(cinrad.io.io.PUP)这部分确实一直都有问题，我也不打算弄了，你有兴趣可以看一下

 

 https://github.com/CyanideCN/PyCINRAD/blob/master/cinrad/io/io.py#L669

laishenggx, 2019-05-16:
好的谢谢

closed 2019-05-16 by laishenggx

CyanideCN, 2019-06-06:
已修复