CyanideCN/PyCINRAD #37: 绘制PUP基本速度产品时单位错误

laishenggx, 2019-07-03:
雷达信息：河池雷达(Z9778)，PUP基本速度产品(V,26)

使用PyCINRAD的PUP接口绘制基本速度时，发现和PUP中的不一致，后来发现是单位没有转换的问题，文件里边数据的单位是节(knots)。

[image]

[image]

CyanideCN, 2019-07-03:
好的，马上修复

CyanideCN, 2019-07-03:
我用其他数据试了一下，并没有出现这个问题
[image]

laishenggx, 2019-07-03:
> 我用其他数据试了一下，并没有出现这个问题

嗯嗯，应该是我们RPG处理的时候还是用的knots，谢谢了

closed 2019-07-03 by laishenggx

CyanideCN, 2019-07-03:
从你那个图来看，我觉得是颜色映射出问题了，有几个色阶被跳过了，连过渡都没有

laishenggx, 2019-07-03:
> 从你那个图来看，我觉得是颜色映射出问题了，有几个色阶被跳过了，连过渡都没有

我试着用metpy.io.level3file的接口试着自己读取并画图了一下，发现应该还是单位的问题，颜色映射(norm)应该是正确的。

没有转换前：
[image]

转换后：
[image]

laishenggx, 2019-07-03:
这个是上图的pup数据文件：
[20170330.093021.02.26.778.txt](https://github.com/CyanideCN/PyCINRAD/files/3353134/20170330.093021.02.26.778.txt)

CyanideCN, 2019-07-03:
好的