CyanideCN/PyCINRAD #39: cinrad.error.RadarDecodeError: Unrecognized data

zhudaoruyi, 2019-08-18:
```python
Cython is not installed, velocity dealias function cannot be used
Traceback (most recent call last):
  File "draft.py", line 61, in <module>
    main()
  File "draft.py", line 56, in main
    draft02()
  File "draft.py", line 45, in draft02
    f = cinrad.io.CinradReader("Z_RADR_I_Z9431_20190723074431_O_DOR_CC_CAP.bin.bz2")
  File "/home/ubuntu/venv35/lib/python3.5/site-packages/cinrad/io/level2.py", line 130, in __init__
    raise RadarDecodeError('Unrecognized data')
cinrad.error.RadarDecodeError: Unrecognized data
```

CyanideCN, 2019-08-18:
麻烦你提供一下报错的数据

zhudaoruyi, 2019-08-19:
[https://pan.baidu.com/s/11meLAVL_gaNG4lewIa_RXA](https://pan.baidu.com/s/11meLAVL_gaNG4lewIa_RXA)

h9k3

CyanideCN, 2019-08-19:
东北的雷达格式似乎和其他地方有点区别，虽然文件名写的是CC雷达，但是格式和CC雷达感觉又不一样，不知道有没有数据格式说明可以参考？

abanger, 2019-09-30:
> 东北的雷达格式似乎和其他地方有点区别，虽然文件名写的是CC雷达，但是格式和CC雷达感觉又不一样，不知道有没有数据格式说明可以参考？

不同地区雷达格式好像还是有差别，如何自己定义，能不能帮大概介绍下，谢谢。

closed 2019-11-22 by CyanideCN