CyanideCN/PyCINRAD #41: 添加转换为pyart.core.Radar类的函数

CyanideCN, 2019-09-06:
转换之后可以方便的利用pyart中的各种计算函数。但是国内双偏振雷达每层径向数不同，不知道如何解决

YvZheng, 2019-09-06:
国内SA/B雷达的径向速度分辨率为250m，反射率为1000m，在pyart的Radar类中，仅提供了一个保存距离的变量range，目前看来暂时无法将国内雷达数据直接转化为Radar类中，除非将反射率插值至250m

CyanideCN, 2019-09-06:
SA/B的格式和nexrad message 1的格式非常像，pyart自己处理的时候是要插值的，也不算很大的问题。
https://github.com/ARM-DOE/pyart/blob/1c2433bf17a2c8316b947e3a84aaa0a6bbfd917e/pyart/io/nexrad_archive.py#L278
Radar这个类的参数实在是太多了，我之前尝试过转换，还是各种报错

bugsuse, 2019-09-10:
如果可以的话，或许我可以协助实现此功能 @CyanideCN

CyanideCN, 2019-09-10:
@bugsuse
非常感谢！我目前设想的是给`cinrad.io.CinradReader`（老版本各种格式的数据）和`cinrad.io.StandardData`（新版本标准格式数据）这两个类分别加自己的bound method。我最近在尝试标准格式的转换，你有兴趣的话可以实现一下CinradReader类的。
PS:可能CinradReader类里面会缺Radar类要求的参数，你可以在`cinrad.io._radar_struct`文件夹里面找到每种雷达对应的dtype然后在CinradReader类里面添加上对应的属性

bugsuse, 2019-09-10:
好的。留个微信吧，交流方便一点。ID：adaynotoday

lovechang1986, 2019-10-01:
这个转换的代码我都写了，但是没有用这个库，由于我不是特别了解雷达，所以仅仅写了基本的参数。稍后用这个库重构一下提交一个pull request，算作一个参考吧。

lovechang1986, 2019-10-01:
如果您自己做建议首先读取一个标准的Radar数据，然后依次检查这个数据里包含的内容，然后根据内容逐步添加即可。

CyanideCN, 2019-10-01:
@lovechang1986 标准数据类的转换我基本已经做好了，老版本数据我现在主要是在纠结插值的问题

lovechang1986, 2019-10-01:
不太懂雷达哈，直接zoom(ref,4)或者repeat4次不行？

wanghui0225, 2019-11-14:
pyart现在可以直接调用啦？如何实现的