CyanideCN/PyCINRAD #45: 读取的azimuth和实际的ray的数目并不一致！

lovechang1986, 2019-10-01:
读取了这个文件：
Z_RADR_I_Z9002_20190930143600_O_DOR_SA_CAP.bin.bz2

```
fn = 'Z_RADR_I_Z9002_20190930143600_O_DOR_SA_CAP.bin.bz2'
udata = CinradReader(fn)
azimuth = udata.azimuth
rays_num = [udata.data.get(i)['REF'].shape[0] for i in udata.data]
print(azimuth)
print(sum(rays_num))
```
azimuth是4404，rays_num的和为4403.
不太一致，请问这是为什么呢？

lovechang1986, 2019-10-01:
把文件共享到百度网盘：
链接: https://pan.baidu.com/s/1JMrBoGyX5iFNG8gUiVo8pw 提取码: sbph 复制这段内容后打开百度网盘手机App，操作更方便哦

CyanideCN, 2019-10-01:
这个算是历史遗留问题吧，一直没时间改，访问方位角应该用`CinradReader.data[tilt]['azimuth']`，CinradReader.azimuth其实我都不太记得是干啥用的了。

用你上传的数据做测试：
```python
>>> sum([f.data[idx]['REF'].shape[0] for idx in range(len(f.el))])
4043
>>> sum([f.data[idx]['azimuth'].shape[0] for idx in range(len(f.el))])
4043
```

lovechang1986, 2019-10-02:
这样说对sa来说放弃了一个Ray的观测数据？可以这样理解吗？

CyanideCN, 2019-10-02:
本来不应该这样的，我不知道为啥最后一组数据不能被读到

closed 2019-10-02 by CyanideCN

CyanideCN, 2019-10-02:
仔细检查了一下，问题出在https://github.com/CyanideCN/PyCINRAD/commit/5a8264c4c13c56ddbe501911faa2cde81f053057
```python
np.append(boundary, data.shape[0] - 1)
```
这个减掉的1就导致了最后一个数据块不会被读到

commit d7d6a600 (2019-10-02): Fix incomplete parsing of SAB files

Close #45