CyanideCN/PyCINRAD #48: CC雷达读取BUG

lovechang1986, 2019-10-14:
在实际测试CC雷达读取中发现该模块存在bug，通过对比MICAPS的结果发现以下问题代码：

`stop_angle = np.where(param["usAngle"] < param["usAngle"][0])[0][0]`
这么判断结束仰角并不合适，从网上搜索的原始C代码可以发现这么来做会更好一些。

`stop_angle = header['ucScanMode'][0] - 100`

CyanideCN, 2019-10-14:
那你提交一个Pull request吧

CyanideCN, 2019-11-22:
我看了一下文档，你的处理方式只适用于VPPI模式，如果是单层PPI就会变成负数。目前来说我还没有碰到过这方面的bug，如果可以的话可否提供一个会出错的数据，我再研究一下。

lovechang1986, 2019-11-27:
出错的原因在于读取的usAngle数组前11个数可能都是单调递增的，但是在12个数的时候并不等于0，如果这个数大于第一个仰角（例如0.5）并且小于其他仰角，那么就会出错了。
下面是读取CC文件错误的详细文件链接，
链接：https://pan.baidu.com/s/1wInKgziZw-ed50vWsupodg 
提取码：36hc 
------
unsigned char ucScanMode;				//扫描方式  1-RHI
//		   10-PPI和ZPPI 
//		   1XX=VPPI(XX为扫描圈数)
......
	ScanPara->num=info.ucScanMode-100;
------
我是参考下面链接的一个文件，因为当时只考虑的VPPI模式，其实可以做判断，根据文件所描述的信息可以做判断，判断是RHI模式或者PPI和ZPPI模式，ucScanMode分别是1或者10.如果是都不是或者超过100的整形，那么就是VPPI。这样会比较完备。因为我当时只是考虑把这个CC雷达转换成pyart的Radar格式，所以没有考虑那么多。
因为我不是专业搞雷达的，我感觉如果是RHI模式或者ZPPI模式可能数据结构又会发生什么变化，和现在的会不一样吧。
链接：https://pan.baidu.com/s/1S6Dv4d3ndmqDf2nTUncvPg 
提取码：ra2c 
------
注：百度网盘有效期7天！
以上内容供参考，我就不提交pull request了，如上所说，我不专业搞雷达，很多原理都不太懂，怕搞错了.

ps：https://github.com/vlouf/eth_radar，这个库是一个计算回波顶高的库，我看使用的是2013年老外的一片最新文献，看是否可加入到你的代码库里。

CyanideCN, 2019-11-28:
你给的这个数据的scanmode是0诶，看起来这个数据像是坏的

CyanideCN, 2019-11-28:
看了一下，是马鞍山雷达啊，这个雷达算是特例吧，格式和其他的都不太一样，我也没想好怎么解决

closed 2020-04-09 by CyanideCN