CyanideCN/PyCINRAD #52: import cinrad时报错

niecj09, 2020-03-20:
import cinrad时报如下错误：
Traceback (most recent call last):
  File "C:/Users/niecq/PycharmProjects/test/read_pup.py", line 5, in <module>
    from cinrad.io import PUP
  File "C:\ProgramData\Miniconda3\lib\site-packages\cinrad-1.5-py3.7.egg\cinrad\__init__.py", line 5, in <module>
    from . import visualize
  File "C:\ProgramData\Miniconda3\lib\site-packages\cinrad-1.5-py3.7.egg\cinrad\visualize\__init__.py", line 8, in <module>
    from cinrad.visualize.ppi import *
  File "C:\ProgramData\Miniconda3\lib\site-packages\cinrad-1.5-py3.7.egg\cinrad\visualize\ppi.py", line 14, in <module>
    from cinrad.visualize.utils import (
  File "C:\ProgramData\Miniconda3\lib\site-packages\cinrad-1.5-py3.7.egg\cinrad\visualize\utils.py", line 124, in <module>
    class ShpReader(shapereader.BasicReader):
AttributeError: module 'cartopy.io.shapereader' has no attribute 'BasicReader'
请问如何解决呢，之前是合适的

zxdawn, 2020-03-20:
@niecj09 cartopy版本是多少？

niecj09, 2020-03-20:
嗯，可能后面cartopy版本自己更新了，看了官网，改了就可以了~
cartopy.io.shapereader.Reader
alias of cartopy.io.shapereader.BasicReader

CyanideCN, 2020-03-20:
cartopy的版本问题有点棘手，之后有空用pyshp替代重写一下

closed 2020-03-20 by CyanideCN