CyanideCN/PyCINRAD #55: 【求助】一个基数据bin文件如何得到等经纬度或者等距离的栅格数据

dreaming-coder, 2020-04-03:
大神，求助啊，基数据我自己能写代码读出来，但是我不是气象专业的，不知道把这多层仰角的数据合成**等高度**   **等经纬度**的栅格数据，实在搞不定，我刚刚才发现正文介绍理由说雷达拼图功能
[image]
请问这个是用来转等经纬度的吗？如果是，能给个示例用法吗？
谢谢！！

CyanideCN, 2020-04-03:
GridMapper是来合并不同雷达的数据的并且无视高度的影响，所以可能不满足你的需求。如果要做CAPPI的话可以参考这个 https://github.com/vlouf/radar_cappi/blob/master/CAPPI.ipynb

CyanideCN, 2020-04-03:
还有一个方法是用kdtree，这样对雷达这种分布不均匀的数据插值比较方便一些

dreaming-coder, 2020-04-03:
好的，谢谢大神，我再看看~~难受啊

dreaming-coder, 2020-04-03:
大神，那我可以对一个雷达基数据文件，不同仰角的数据，利用GridMapper来合并吗？
然后一个雷达基数据文件的数据就变成三维的(方向角，极径，dBZ)的数据了，这时候利用kdTree来查找栅格点最近的有数据的值进行插值，来得出栅格网络？

CyanideCN, 2020-04-05:
计算三维的(方向角，极径，dBZ)的数据不用GridMapper都可以实现，具体参见
https://github.com/CyanideCN/PyCINRAD/blob/71029a4478f085e799d8470e99149e2ce15e3280/cinrad/easycalc.py#L33-L47

```python
r_list = list(f.iter_tilt(230, 'REF'))
data, distance, azimuth, elevation = cinrad.easycalc._extract(r_list)
```

dreaming-coder, 2020-04-05:
我想到了，我先用cinrad，得到每个数据的经纬度和高度，然后用的scipy的griddata来进行三维空间的插值，谢谢大佬~~^_^

CyanideCN, 2020-04-05:
griddata容易炸内存哦，慎用

dreaming-coder, 2020-04-05:
就用nearest的模式，cubic没法针对三维的，linear又太生硬，我16G试试看，抱有侥幸~~
对了，大神，你贴的代码的意思是不是遍历所有仰角的数据，返回dbz，该点到雷达站的径长，方向角，仰角这四个列表，每次取数据用同一个下标来取，然后利用球坐标和直角坐标转换得到直角坐标~但是还是要插值啊，插值我百度发现好多用的griddata，好像还是要用内存

CyanideCN, 2020-04-05:
是的，我刚刚贴的代码只是获得3D的array。用xarray.DataArray的interp方法应该也可以插值CAPPI，可以把这个三维数据变成DataArray然后插值试试，应该没有griddata那么痛苦。
http://xarray.pydata.org/en/stable/generated/xarray.DataArray.interp.html

dreaming-coder, 2020-04-05:
好的，大神辛苦了，谢谢，我这几天真被老师逼疯了，多亏大神救命啊

closed 2020-06-17 by CyanideCN