CyanideCN/PyCINRAD #64: 能处理双偏振雷达产品数据吗？

wangyongtao0929, 2020-07-01:
[雷达0623.zip](https://github.com/CyanideCN/PyCINRAD/files/4856289/0623.zip)

CyanideCN, 2020-07-01:
```python
>>> import cinrad
>>> f = cinrad.io.level3.StandardPUP('Z9571_20200619092314Z_PPI_01_19')
>>> r = f.get_data()
>>> r
<xarray.Dataset>
Dimensions:    (azimuth: 365, distance: 230)
Coordinates:
  * azimuth    (azimuth) float32 6.280654 0.0145735 ... 6.24601 6.2632008
  * distance   (distance) float64 0.0 1.0 2.0 3.0 ... 226.0 227.0 228.0 229.0
Data variables:
    REF        (azimuth, distance) float64 nan nan nan -6.0 ... nan nan nan nan
    longitude  (azimuth, distance) float64 120.3 120.3 120.3 ... 120.3 120.3
    latitude   (azimuth, distance) float64 30.27 30.28 30.29 ... 32.33 32.34
    height     (azimuth, distance) float64 0.096 0.1045 0.1131 ... 5.077 5.113
Attributes:
    elevation:        0.48339844
    range:            230.0
    scan_time:        2020-06-19 09:23:14
    site_code:        Z9571
    site_name:        杭州
    site_longitude:   120.3375
    site_latitude:    30.273333
    tangential_reso:  1.0
    task:             VCP21D
>>>
```

closed 2020-07-01 by CyanideCN

CyanideCN, 2020-07-01:
记得从github拉最新版本，我刚刚提交了一些改动