CyanideCN/PyCINRAD #65: 我也遇到了cinrad.calc.VCS(ref)没有任何反应的情况

J1alin, 2020-07-02:
我安装的cinrad的版本是1.6的。
数据读取后画PPI图是正常的，但是画剖面图的时候，
ref = [f.get_data(i, 230, 'REF') for i in f.angleindex_r]
vcs = cinrad.calc.VCS(ref)
sec = vcs.get_section(start_cart=(113.4, 23.1), end_cart=(113.5, 23.2))
print(sec)
fig = cinrad.visualize.PPI(ref[0])
fig.plot_cross_section(sec)
fig('D:\CINRAD_SA_Data')
到了第二行那里就没有任何反应了，也没有错误提示信息，也没有图出来，很久也不会自动停下来。

CyanideCN, 2020-07-02:
我这里无法复现，你可以用master分支再测试一下吗

J1alin, 2020-07-02:
谢谢您的及时回复。

我是初学者，所以您说的master分支测试我不是很明白，还需要学习。

下面是代码，PPI图是可以正常显示的，就是剖面不行。（pycharm）
import cinrad
from cinrad.visualize import Section
from cinrad.easycalc import GridMapper  # 导入库函数
sa_radar_file = 'D:/CINRAD_SA_Data/Z_RADR_I_Z9200_20190414065401_O_DOR_SAD_CAP_FMT.bin.bz2'
from cinrad.io import CinradReader, StandardData
f = StandardData(sa_radar_file)
print(sa_radar_file)
# 将鼠标点到get_data处然后ctrl+b 即可看对应函数定义
ref = f.get_data(0, 230, 'REF')
print(ref)
fig = cinrad.visualize.PPI(ref, dpi=600, style='white')
fig('D:\CINRAD_SA_Data')ref = [f.get_data(i, 230, 'REF') for i in f.angleindex_r]
vcs = cinrad.calc.VCS(ref)
sec = vcs.get_section(start_cart=(113.4, 23.1), end_cart=(113.5, 23.2))
print(sec)
fig = cinrad.visualize.PPI(ref[0])
fig.plot_cross_section(sec)
fig('D:\CINRAD_SA_Data')
折腾了两天了，把所有的东西在不同的机器上都试过了，也是一样的问题。很奇怪。

谢谢您。

CyanideCN, 2020-07-02:
你在github下载最新的版本（网页右边有一个绿色的下载按钮）然后重新安装，再试试

J1alin, 2020-07-02:
好的，非常感谢。------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;&quot;CyanideCN&quot;<notifications@github.com&gt;
发送时间:&nbsp;2020年7月2日(星期四) 晚上9:09
收件人:&nbsp;&quot;CyanideCN/PyCINRAD&quot;<PyCINRAD@noreply.github.com&gt;;
抄送:&nbsp;&quot;J1alin&quot;<1209960968@qq.com&gt;;&quot;Author&quot;<author@noreply.github.com&gt;;
主题:&nbsp;Re: [CyanideCN/PyCINRAD] 我也遇到了cinrad.calc.VCS(ref)没有任何反应的情况 (#65)

J1alin, 2020-07-03:
已经解决了，好像没什么问题。只是速度超级慢。要等很久才行。

closed 2020-07-03 by J1alin