CyanideCN/PyCINRAD #66: PUP文件读取错误求助

boyucjz, 2020-08-02:
from cinrad.io import PUP    这一句可以正常执行
f=PUP('Z_20200727094405Z_CAPPI_00_110')   执行一句时总是报错， 在python3.7  python3.8 都试了

报错信息如下：
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): M
ax retries exceeded with url: /Unidata/MetPy/v0.12.1/staticdata/sfstns.tbl (Caused by NewConnectionE
rror('<urllib3.connection.HTTPSConnection object at 0x0000000014F19248>: Failed to establish a new c
onnection: [Errno 11004] getaddrinfo failed'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\Python37\lib\site-packages\cinrad\io\level3.py", line 41, in __init__
    from metpy.io.nexrad import Level3File
  File "D:\Python37\lib\site-packages\metpy\io\__init__.py", line 12, in <module>
    from .metar import *  # noqa: F403
  File "D:\Python37\lib\site-packages\metpy\io\metar.py", line 15, in <module>
    from .station_data import station_info
  File "D:\Python37\lib\site-packages\metpy\io\station_data.py", line 140, in <module>
    station_info = StationLookup()
  File "D:\Python37\lib\site-packages\metpy\io\station_data.py", line 128, in __init__
    self._sources = [dict(_read_station_table()), dict(_read_master_text_file()),
  File "D:\Python37\lib\site-packages\metpy\io\station_data.py", line 35, in _read_station_table
    input_file = get_test_data('sfstns.tbl', as_file_obj=False)
  File "D:\Python37\lib\site-packages\metpy\cbook.py", line 33, in get_test_data
    path = POOCH.fetch(fname)
  File "D:\Python37\lib\site-packages\pooch\core.py", line 579, in fetch
    stream_download(url, full_path, known_hash, downloader, pooch=self)
  File "D:\Python37\lib\site-packages\pooch\core.py", line 747, in stream_download
    downloader(url, tmp, pooch)
  File "D:\Python37\lib\site-packages\pooch\downloaders.py", line 167, in __call__
    response = requests.get(url, **kwargs)
  File "D:\Python37\lib\site-packages\requests\api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "D:\Python37\lib\site-packages\requests\api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "D:\Python37\lib\site-packages\requests\sessions.py", line 530, in request
    resp = self.send(prep, **send_kwargs)
  File "D:\Python37\lib\site-packages\requests\sessions.py", line 665, in send
    history = [resp for resp in gen]
  File "D:\Python37\lib\site-packages\requests\sessions.py", line 665, in <listcomp>
    history = [resp for resp in gen]
  File "D:\Python37\lib\site-packages\requests\sessions.py", line 245, in resolve_redirects
    **adapter_kwargs
  File "D:\Python37\lib\site-packages\requests\sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
  File "D:\Python37\lib\site-packages\requests\adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443)
: Max retries exceeded with url: /Unidata/MetPy/v0.12.1/staticdata/sfstns.tbl (Caused by NewConnecti
onError('<urllib3.connection.HTTPSConnection object at 0x0000000014F19248>: Failed to establish a ne
w connection: [Errno 11004] getaddrinfo failed'))
>>>

boyucjz, 2020-08-02:
请问是数据文件格式不对，还是没有安装好？

CyanideCN, 2020-08-02:
模块是用的metpy来解PUP文件，在第一次解码的时候metpy会下载一些数据到本地，如果网络不好的话就会出现这个问题

CyanideCN, 2020-08-02:
而且你这个数据好像是新版的PUP数据，CAPPI产品目前还不支持读取

boyucjz, 2020-08-05:
多谢，按照  https://www.cnblogs.com/zuoanfengxi/p/12724698.html   这个方法解决了网络下载问题；文件还是打不开，像你说的，文件格式好像是新格式。

closed 2020-08-05 by boyucjz