CyanideCN/PyCINRAD #67: 产品号37（组合反射率） 文件读取错误求助

boyucjz, 2020-08-14:
D:\Python37>python.exe
Python 3.7.8 (tags/v3.7.8:4b47a5b6ba, Jun 28 2020, 08:53:46) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import cinrad
>>> f = cinrad.io.level3.StandardPUP('Z_20200727093842Z_CR_00_37')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\Python37\lib\site-packages\cinrad\io\level3.py", line 428, in __init__
    self._parse()
  File "D:\Python37\lib\site-packages\cinrad\io\level3.py", line 456, in _parse
    params = get_product_param(ptype, self.f.read(64))
  File "D:\Python37\lib\site-packages\cinrad\io\level3.py", line 415, in get_product_param
    return _ProductParams(ptype, param_bytes).params
  File "D:\Python37\lib\site-packages\cinrad\io\level3.py", line 398, in __init__
    map_func[ptype]()
KeyError: 18
>>>

文件如下：
[Z_20200727093842Z_CR_00_37.zip](https://github.com/CyanideCN/PyCINRAD/files/5073440/Z_20200727093842Z_CR_00_37.zip)

CyanideCN, 2020-08-14:
组合反射率数据暂时还不支持，之后会添加的------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;&quot;boyucjz&quot;<notifications@github.com&gt;
发送时间:&nbsp;2020年8月14日(星期五) 下午4:30
收件人:&nbsp;&quot;CyanideCN/PyCINRAD&quot;<PyCINRAD@noreply.github.com&gt;;
抄送:&nbsp;&quot;Subscribed&quot;<subscribed@noreply.github.com&gt;;
主题:&nbsp;[CyanideCN/PyCINRAD] 产品号37（组合反射率） 文件读取错误求助 (#67)

CyanideCN, 2020-08-14:
我在数据说明里面没有找到产品类型为18的产品，你那里有新一点的数据说明吗。
[image]

boyucjz, 2020-08-15:
我也没有最新的格式说明，只能在网上找到这个，确实找不到18对应的说明
https://www.doc88.com/p-00787765029520.html

hehuan710, 2020-10-03:
我的数据也是KeyError: 18
[Z9716_20201003122400Z_CR_00_37.zip](https://github.com/CyanideCN/PyCINRAD/files/5321888/Z9716_20201003122400Z_CR_00_37.zip)

boyucjz, 2020-10-22:
> 我的数据也是KeyError: 18 强制改为1，貌似也不支持读取格点类型
> [Z9716_20201003122400Z_CR_00_37.zip](https://github.com/CyanideCN/PyCINRAD/files/5321888/Z9716_20201003122400Z_CR_00_37.zip)

boyucjz, 2020-10-22:
> 我的数据也是KeyError: 18 强制改为1，貌似也不支持读取格点类型
> [Z9716_20201003122400Z_CR_00_37.zip](https://github.com/CyanideCN/PyCINRAD/files/5321888/Z9716_20201003122400Z_CR_00_37.zip)

强制改为1可以使用吗？

hehuan710, 2020-10-22:
> > 我的数据也是KeyError: 18 强制改为1，貌似也不支持读取格点类型
> > [Z9716_20201003122400Z_CR_00_37.zip](https://github.com/CyanideCN/PyCINRAD/files/5321888/Z9716_20201003122400Z_CR_00_37.zip)
> 
> 强制改为1可以使用吗？

不行哦  好像现在还不支持读取格点类型数据

qianz98, 2021-06-19:
请问，现在可以支持读取格点类型数据了吗？

hehuan710, 2021-06-20:
> 请问，现在可以支持读取格点类型数据了吗？

貌似还不行

yejia414, 2023-02-13:
类型为18的产品类型还是没办法解析数据么？

yejia414, 2023-02-14:
@boyucjz 你好，我这边产品类型为18的文件用栅格格式可以解析，Z_20200727093842Z_CR_00_37.zip解析后生成图片如下[
[image]
](url)

CyanideCN, 2023-04-21:
最新commit已经添加支持

closed 2023-04-21 by CyanideCN