CyanideCN/PyCINRAD #70: 双偏振27号速度产品数据读取错误及ppi绘图问题

hehuan710, 2020-09-25:
两个问题求助：
1、双偏振27号速度产品数据读取错误 ，其他产品读取正常
f=cinrad.io.level3.StandardPUP('Z9716_20200925083000Z_PPI_01_27')
data = f.get_data()
ValueError:Could not convert tuple of form (dims, data[, attrs, encoding]): (['azimuth', 'distance'], masked_array(data=[--, 2.0, -14.0, -9.5, -5.5, 8.5, 3.5, -11.5, -7.5............
[Z9716_20200925083000Z_PPI_01_27.zip](https://github.com/CyanideCN/PyCINRAD/files/5281672/Z9716_20200925083000Z_PPI_01_27.zip)
2、ppi绘制的图片0°方向有缺失
f=cinrad.io.level3.StandardPUP('Z9716_20200909100600Z_PPI_01_158')
data = f.get_data()
fig = PPI(data ,dpi=150, add_city_names=True,plot_labels=True,style='black')
[Z9716_20200909100600Z_PPI_01_158.zip](https://github.com/CyanideCN/PyCINRAD/files/5281696/Z9716_20200909100600Z_PPI_01_158.zip)
[image]

CyanideCN, 2020-09-27:
第二个问题已经在 71bccb9fd02f30470fa535d48a1ab609edba5b84 里面修复了

CyanideCN, 2020-09-29:
已修复

closed 2020-09-29 by CyanideCN

hehuan710, 2020-10-01:
> 已修复

那RF就没有了

CyanideCN, 2020-10-02:
改了一下，现在应该没有问题了

hehuan710, 2020-10-02:
> 改了一下，现在应该没有问题了

感谢，还有个小问题，不知怎么回事：读取出来的164分类产品和pup软件对比，发现没有“0”的分类，就是图中绿色的部分  
PS：这个Z9716_20201002005400Z_HCL_01_164文件貌似有点问题，dtype为11，貌似14才对，ptype是51
[image]
[image]
[Z9716_20201002005400Z_HCL_01_164.zip](https://github.com/CyanideCN/PyCINRAD/files/5316681/Z9716_20201002005400Z_HCL_01_164.zip)

CyanideCN, 2020-10-03:
已修复

commit ed911999 (2020-09-29): Fix #70