CyanideCN/PyCINRAD #76: 站点缺经纬度求助

danchunxiaokeai, 2020-12-19:
[image]
[image]
[image]
老师不好意思打扰了，我最近在用cinrad画雷达。出现反射率画出来了以及时间输出也对，但是站点经纬度没有的问题。我猜测可能是cinrad里一些小站点或者新站点的经纬度没有收录。想请教一下是否我可以自己修改源代码的站点经纬度设置可能可以解决这个问题。

CyanideCN, 2020-12-19:
我之后会把站点数据换成json格式，这样方便编辑一些，目前的话你可以用f.stationlat = ... 和f.stationlon = ... 来手动设置

danchunxiaokeai, 2020-12-19:
谢谢老师的解答！祝好！！

commit 613cbb98 (2020-12-21): Change radar station info to json, fix #76
closed 2020-12-21 by CyanideCN