CyanideCN/PyCINRAD #78: SA格式雷达基数据读取错误求助

jinhuanz, 2021-04-19:
强大的雷达数据处理工具，三行命令读取数据，还能方便地转成nc格式👍  

我这边PUP产品数据读取没有问题，处理基数据遇到如下问题：
In [9]: CinradReader(your_radar_file)
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-9-54d6b988acb5> in <module>
----> 1 CinradReader(your_radar_file)

~/miniconda3/envs/radar/lib/python3.7/site-packages/cinrad-1.6.3-py3.7.egg/cinrad/io/level2.py in __init__(self, file, radar_type)
    102         f.seek(0)
    103         if self.radartype in ["SA", "SB"]:
--> 104             self._SAB_reader(f)
    105         elif self.radartype in ["CA", "CB"]:
    106             self._SAB_reader(f, dtype="CAB")

~/miniconda3/envs/radar/lib/python3.7/site-packages/cinrad-1.6.3-py3.7.egg/cinrad/io/level2.py in _SAB_reader(self, f, dtype)
    150             radar_dtype = S_SPECIAL_dtype
    151             _header_size = 132
--> 152         data = np.frombuffer(f.read(), dtype=radar_dtype)
    153         start = datetime.datetime(1969, 12, 31)
    154         deltday = datetime.timedelta(days=int(data["day"][0]))

ValueError: buffer size must be a multiple of element size

数据文件： 
[Z_RADR_I_Z9220_20210414114800_P_DOR_SA_V_5_115_15.220.bin.zip](https://github.com/CyanideCN/PyCINRAD/files/6334154/Z_RADR_I_Z9220_20210414114800_P_DOR_SA_V_5_115_15.220.bin.zip)

jinhuanz, 2021-04-19:
https://github.com/CyanideCN/PyCINRAD/issues/77

closed 2021-04-19 by jinhuanz

CyanideCN, 2021-04-19:
你这个还是PUP产品吧
```
<xarray.Dataset>
Dimensions:    (azimuth: 364, distance: 230)
Coordinates:
  * azimuth    (azimuth) float64 0.001745 0.01905 0.03636 ... 6.268 0.001745
  * distance   (distance) float64 0.5 1.0 1.5 2.0 ... 113.5 114.0 114.5 115.0
Data variables:
    VEL        (azimuth, distance) float64 nan nan nan 4.5 ... nan nan nan nan
    longitude  (azimuth, distance) float64 117.7 117.7 117.7 ... 117.7 117.7
    latitude   (azimuth, distance) float64 39.05 39.05 39.06 ... 40.07 40.08
    RF         (azimuth, distance) float64 nan nan nan nan ... nan nan nan nan
Attributes:
    elevation:        1.5
    range:            115
    scan_time:        2021-04-14 11:48:00
    site_code:        Z9220
    site_name:        塘沽
    site_longitude:   117.718
    site_latitude:    39.043
    tangential_reso:  0.5
```

jinhuanz, 2021-04-19:
emmm....  看起来是的... 
雷达数据小白 :(
我以为R、V、W三个都是基数据

jinhuanz, 2021-04-19:
更新了问题

jinhuanz, 2021-04-19:
https://github.com/CyanideCN/PyCINRAD/issues/53