CyanideCN/PyCINRAD #86: GridMapper做雷达拼图，无法显示

tooleek, 2022-09-26:
`for binFile in binFileList:
    f = StandardData(binFile)
    refData = f.get_data(0, drange, 'REF')
    allData.append(refData)

gm = GridMapper(allData)

grid = gm(0.1)
fig = PPI(grid, dpi=75, add_city_names=False)
fig._save(outPath)`

请问，使用该方法做雷达拼图，但是最终显示的图缺少雷达数据，是什么原因呢

CyanideCN, 2022-09-27:
代码看起来没问题，确定基数据里是有回波的吗

tooleek, 2022-09-28:
嗯，基数据是有回波的

CyanideCN, 2022-09-29:
具体拼图画出来是啥样的

tooleek, 2022-09-29:
**三个站的单站图如下：**
[image]
[image]
[image]
**合成后的图如下：**
[image]

**代码如下：**
binFileList = binFiles.split(",")

allData = []

for binFile in binFileList:
    f = StandardData(binFile)
    refData = f.get_data(0, drange, 'REF')
    # fig = PPI(refData, dpi=75, add_city_names=False)
    # fig._save(outPath)
    allData.append(refData)

gm = GridMapper(allData)

grid = gm(0.1)
fig = PPI(grid, dpi=75, add_city_names=False)
fig._save(outPath)
print("【雷达拼图】输出成功")

**执行过程中，抛了如下异常：**
D:\Software\Anaconda3\lib\site-packages\cinrad\calc.py:425: RuntimeWarning: divide by zero encountered in true_divide
  return np.ma.average(inp, weights=1 / wgt, axis=2)
D:\Software\Anaconda3\lib\site-packages\numpy\ma\extras.py:621: RuntimeWarning: invalid value encountered in multiply
  wgt = wgt*(~a.mask)
D:\Software\Anaconda3\lib\site-packages\cinrad\visualize\ppi.py:233: RuntimeWarning: All-NaN slice encountered
  "Max: {:.1f}{}".format(np.nanmax(var), unit[self.dtype]),

tooleek, 2022-09-29:
我把参数调成  grid = gm(0.001)，就有数据了，看来是0.1太小了这个原因导致的

tooleek, 2022-09-30:
现在有一个新的问题，就是组合反射率的雷达拼图，要抛异常，找不到 distance

CyanideCN, 2022-09-30:
组合反射率目前不支持拼图

muyi151, 2022-10-23:
我也是类似的问题，“file2 = '20190516.000133.01.19.552'   #旧pup
f2 = PUP(file2)
data2 = f2.get_data()
gm=cinrad.calc.GridMapper([data0,data2])
grid = gm(0.1)”前面一步没有报错，最后一步“grid = gm(0.1)”却会报错“  File "/home/laoping/anaconda3/envs/cinrad/lib/python3.8/site-packages/cinrad/calc.py", line 455, in __call__
    r_attr["nyquist_vel"],
KeyError: 'nyquist_vel'”请问这是为什么？

CyanideCN, 2022-11-02:
> 我也是类似的问题，“file2 = '20190516.000133.01.19.552' #旧pup f2 = PUP(file2) data2 = f2.get_data() gm=cinrad.calc.GridMapper([data0,data2]) grid = gm(0.1)”前面一步没有报错，最后一步“grid = gm(0.1)”却会报错“ File [...]

@muyi151 这个算是一个bug，后续会修复

2356852131, 2023-03-03:
[image]
[image]

closed 2025-03-04 by CyanideCN