CyanideCN/PyCINRAD #90: 中国气象台最新下发的双偏振雷达数据，无法读取数据，大佬能帮忙看看什么原因吗，万分感谢

yuunl, 2023-03-08:
- 错误信息
`Traceback (most recent call last):
  File "D:\ANDCONDA\envs\product2cinrad\lib\site-packages\cinrad\io\level2.py", line 545, in __init__
    self._parse()
  File "D:\ANDCONDA\envs\product2cinrad\lib\site-packages\cinrad\io\level2.py", line 635, in _parse
    data_body = np.frombuffer(
TypeError: data type 'u0' not understood
python-BaseException
[[`]][image]
[][image]
- 雷达数据
[Radar.zip](https://github.com/CyanideCN/PyCINRAD/files/10919200/Radar.zip)

yuunl, 2023-03-08:
使用的是新数据的读取方法
`f = cinrad.io.StandardData('D:\data\Radar\Z_RADR_I_Z9411_20230308080400_P_DOR_SAD_CAR_250_230_NUL_FMT.bin.bz2')`

yuunl, 2023-03-08:
[image]

CyanideCN, 2023-03-09:
你这个并不是基数据，不能用`StandardData`
```python
>>> f = cinrad.io.level3.StandardPUP('Z_RADR_I_Z9411_20230308080400_P_DOR_SAD_V_500_115_15_FMT.bin.bz2')
>>> data = f.get_data()
>>> print(data)
<xarray.Dataset>
Dimensions:    (azimuth: 359, distance: 230)
Coordinates:
  * azimuth    (azimuth) float64 6.28 0.0145 0.03205 0.0496 ... 6.245 6.263 6.28
  * distance   (distance) float64 0.5 1.0 1.5 2.0 ... 113.5 114.0 114.5 115.0
Data variables:
    VEL        (azimuth, distance) float64 nan nan -2.5 0.0 ... nan nan nan nan
    longitude  (azimuth, distance) float64 121.6 121.6 121.6 ... 121.6 121.6
    latitude   (azimuth, distance) float64 38.91 38.92 38.92 ... 39.94 39.94
    height     (azimuth, distance) float64 0.1791 0.1922 0.2054 ... 3.934 3.954
    RF         (azimuth, distance) float64 nan nan nan nan ... nan nan nan nan
Attributes:
    elevation:        1.5
    range:            115.0
    scan_time:        2023-03-08 08:04:11
    site_code:        Z9411
    site_name:        大连
    site_longitude:   121.648056
    site_latitude:    38.908054
    tangential_reso:  0.5
    task:             VCP21D
```
目前`StandardPUP`只支持一部分ROSE产品，格点化的数据都是无法读的

closed 2023-03-09 by CyanideCN

yuunl, 2023-03-09:
大佬，还有个问题想请教一下，pup产品解出来的数据能否转为基数据文件啊

Lily2069, 2024-04-07:
您好，请问rose2.0的数据格式文档可以分享一下吗

edwardlikaiyuan, 2024-04-10:
> 您好，请问rose2.0的数据格式文档可以分享一下吗

http://www.cmastd.cn/standardView.jspx?id=4131      单站雷达基数据及产品格式 
标准编号 | QX/T 653-2022