CyanideCN/PyCINRAD #91: 不同版本绘制剖面有明显差别问题

lingyunfeng, 2023-04-19:
发现主要是1.4.1版本与1.4.2、1.5.1、1.6.1、1.7.2等版本绘制的剖面明显要完整很多，不知道1.4.2之后版本是哪些代码有改导致的，请检查下。1.6.1是否可以绘制的剖面和1.4.1一致？
[Uploading Z_RADR_I_Z9072_20180527100300_O_DOR_SA_CAP.bin.bz2…]()
[Uploading PYCINRAD不同版本垂直剖面VCS差异.docx…]()

lingyunfeng, 2023-04-19:
1.4.1
[image]

1.4.2
[image]

1.6.1
[image]

1.7.2
[image]

lingyunfeng, 2023-04-19:
剖面起始位置
    start_cart=(109.20, 19.09)
    end_cart=(109.89, 19.64)
    spacing=500

CyanideCN, 2023-04-20:
翻了一下，1.4.1和1.4.2的区别大概在 cc88c9b2b33d61cf89e1824db30acee75391e111 里
```diff
-        r[np.isnan(r)] = 0
+        r = np.ma.masked_invalid(r)
```
之前的剖面只支持反射率，所以把没有值的地方填成0也不影响，但是后续加上了其他要素的剖面，填0就不行了。这个改动应该会影响`contourf`的逻辑。

lingyunfeng, 2023-04-23:
谢谢大牛，我再试试