CyanideCN/PyCINRAD #95: 组合反射率的数据如何变成格点数据

liyuan3970, 2023-08-29:
您好，我用您的算法出来的组合反射率的数据自己用contour画图，回波地理位置存在偏移的情况，
path ="/home/workspace/Data/My_Git/web_met/static/data/downfile/" 
f = cinrad.io.StandardData(path+'Z_RADR_I_Z9576_20230822032019_O_DOR_SAD_CAP_FMT.bin.bz2')
tilt_number = 0
data_radius = 230
data_dtype = 'REF' # stands for reflectivity
# 数据加载核心
ra = f.get_data(tilt_number, data_radius, data_dtype)
rl = list(f.iter_tilt(230, 'REF'))
cr = cinrad.calc.quick_cr(rl)
data = cr.CR
## 颜色 ------
colorslist = ['#00C800','#019000','#FFFF00','#E7C000','#FF9000','#D60000','#C00000','#FF00F0','#780084','#AD90F0','#AE0AF5']
levels1 = [20,25,30,35,40,45,50,55,60,65,70,75]
cmaps = LinearSegmentedColormap.from_list('mylist',colorslist,N=11)
contour = plt.contourf(data.longitude, data.latitude, data,cmap=cmaps,levels = levels1)

后来我验证了用单仰角绘制最低仰角的反射率是不存在位置偏移的情况的
path ="/home/workspace/Data/My_Git/web_met/static/data/downfile/" 
f = cinrad.io.StandardData(path+'Z_RADR_I_Z9576_20230822032019_O_DOR_SAD_CAP_FMT.bin.bz2')
v = f.get_data(0,230, 'REF')
fg =cinrad.calc.GridMapper([v])
output = fg(0.01)
colorslist = ['#00C800','#019000','#FFFF00','#E7C000','#FF9000','#D60000','#C00000','#FF00F0','#780084','#AD90F0','#AE0AF5']
levels1 = [20,25,30,35,40,45,50,55,60,65,70,75]
cmaps = LinearSegmentedColormap.from_list('mylist',colorslist,N=11)
contour = plt.contourf(output.longitude, output.latitude, output.REF,cmap=cmaps,levels = levels1)

请问您这个包有组合反射率数据变成格点数据的方法么？或者调用您底层函数实现这一点您有什么建议?

CyanideCN, 2023-08-30:
组合反射率有一点偏移很正常啊

hehuan710, 2023-09-28:
> 组合反射率有一点偏移很正常啊
> […](#)

[image]
[image]
从边界上看0.5和组合反射率确实有比较明显的偏移

hehuan710, 2023-09-28:
> 组合反射率有一点偏移很正常啊
> […](#)
[image]
[image]
从地图边界来看，组合反射率确实较0.5°有明显偏差，PS：这样比较是因为0.5°以上仰角基本没回波了

hehuan710, 2023-09-29:
> 组合反射率有一点偏移很正常啊
> […](#)

[image]
原因可能在此处，因为每个仰角的经纬度范围有差异，导致CR偏移

liyuan3970, 2023-09-29:
我回头看看，这个0.5仰角的位置偏移不明显，如果这里有问题，那经纬度按照0.5仰角的来应该来解决

CyanideCN, 2023-10-08:
可以看看这个commit有没有修复，刚刚改了一下