CyanideCN/PyCINRAD #96: 菜鸡礼貌的问一句关于Cinrad读取X波段相控阵雷达数据的问题

December0w0, 2023-09-05:
`我在做广东那边X波段雷达资料的解码，目前想要把雷达资料形式转换成pyart可以读取的形式。我使用了CinradReader和StandardData尝试读取资料。但CinradReader报错为附录1，StandardData的报错信息为附录2，并且我确定错误是出现在读取资料处。我很好奇为什么会出现这种情况，是cinrad没法识别广东那边的数据吗（悲伤）

**附录1——CinradReader报错信息**
`Traceback (most recent call last):

  File D:\Anaconda3\Lib\site-packages\cinrad\io\level2.py:125 in __init__
    self._SAB_reader(f, dtype="special")

  File D:\Anaconda3\Lib\site-packages\cinrad\io\level2.py:154 in _SAB_reader
    data = np.frombuffer(f.read(), dtype=radar_dtype)

ValueError: buffer size must be a multiple of element size

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File D:\Anaconda3\Lib\site-packages\spyder_kernels\py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File e:\代码\python代码\radar_qc_outputcfradial（cinrad）.py:38
    PRD = CinradReader(os.path.join(radar_dir+'/'+os.path.basename(_)))  #CinradReader   StandardData

  File D:\Anaconda3\Lib\site-packages\cinrad\io\level2.py:127 in __init__
    raise err

  File D:\Anaconda3\Lib\site-packages\cinrad\io\level2.py:118 in __init__
    raise RadarDecodeError("Unrecognized data")

RadarDecodeError: Unrecognized data
`
**附录2 StandardData报错信息**
`Traceback (most recent call last):

  File D:\Anaconda3\Lib\site-packages\spyder_kernels\py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File e:\代码\python代码\radar_qc_outputcfradial（cinrad）.py:38
    PRD = StandardData(os.path.join(radar_dir+'/'+os.path.basename(_)))  #CinradReader   StandardData

  File D:\Anaconda3\Lib\site-packages\cinrad\io\level2.py:546 in __init__
    self._parse()

  File D:\Anaconda3\Lib\site-packages\cinrad\io\level2.py:636 in _parse
    data_body = np.frombuffer(

TypeError: data type 'u1028' not understood
`

CyanideCN, 2023-09-05:
X波段的话试一下`cinrad.io.PhasedArrayData`

December0w0, 2023-09-06:
> X波段的话试一下`cinrad.io.PhasedArrayData`

解决啦，感谢！

closed 2023-09-08 by CyanideCN

Bcupb, 2024-01-12:
> > X波段的话试一下`cinrad.io.PhasedArrayData`
> 
> 解决啦，感谢！

您好，请问一下您是怎么解决的？万分感谢

December0w0, 2024-01-26:
> > > X波段的话试一下`cinrad.io.PhasedArrayData`
> > 
> > 
> > 解决啦，感谢！
> 
> 您好，请问一下您是怎么解决的？万分感谢

可以礼貌问一下你用的是什么型号的雷达吗？

yinliang111111111, 2024-11-13:
> X波段的话试一下`cinrad.io.PhasedArrayData`

请问使用这个读取出现以下报错，这个是什么问题啊：Traceback (most recent call last):
  File "D:\研究生文件\研一\光流法目标数据\PPIangPHI\main.py", line 48, in <module>
    reader = cinrad.io.PhasedArrayData(file_path)
  File "D:\研究生文件\研一\光流法目标数据\PPIangPHI\venv\lib\site-packages\cinrad\io\level2.py", line 928, in __init__
    self._d = data = np.frombuffer(f.read(), dtype=PA_radial)
ValueError: buffer size must be a multiple of element size