CyanideCN/PyCINRAD #104: 部分CC波段站点旧格式基数据读取失败

Notte118, 2024-03-14:
版本：1.8.0

脚本：
from cinrad.io import CinradReader

path = r"Z_RADR_I_Z9438_20170904160044_O_DOR_CC_CAP.bin.bz2"  # 松原站 2017年旧格式
CinradReader(path)

错误信息：
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    CinradReader(path, "CC")
  File "env\lib\site-packages\cinrad\io\level2.py", line 127, in __init__
    raise err
  File "env\lib\site-packages\cinrad\io\level2.py", line 114, in __init__
    self._CC_reader(f)
  File "env\lib\site-packages\cinrad\io\level2.py", line 218, in _CC_reader
    raise NotImplementedError("Only VPPI scan mode is supported")
NotImplementedError: Only VPPI scan mode is supported

查看源码，在 _CC_reader() 方法中报错语句：

  scan_mode = header["ucScanMode"][0]
  if scan_mode < 100:
      raise NotImplementedError("Only VPPI scan mode is supported")
  stop_angle = scan_mode - 100
  self.scantime = (
      datetime.datetime(
          header["ucEYear1"][0] * 100 + header["ucEYear2"][0],
          header["ucEMonth"][0],
          header["ucEDay"][0],
          header["ucEHour"][0],
          header["ucEMinute"][0],
          header["ucESecond"][0],
      )
      - utc_offset
  )
    
这里 scan_mode 是 61 导致了报错。
强制修改该值继续运行，会在提取 self.scantime 时，出现年份为0的报错。

请问这种情况如何解决，疑似解码问题.

CyanideCN, 2024-03-14:
试一下把格式指定为SA，不行的话再试试CA

Notte118, 2024-03-15:
> 试一下把格式指定为SA，不行的话再试试CA

经过尝试，主动选用这两种波段读取，均会调用 _SAB_reader() 方法，在 data = np.frombuffer(f.read(), dtype=radar_dtype) 这一句中，使用 SAB、CAB 对应的 radar_dtype (numpy.dtypes.VoidDType对象)限定读取格式，均会报错 ValueError: buffer size must be a multiple of element size. 格式可能不太匹配

错误信息：
  File "env\lib\site-packages\cinrad\io\level2.py", line 108, in __init__
    self._SAB_reader(f, dtype="CAB")
  File "env\lib\site-packages\cinrad\io\level2.py", line 154, in _SAB_reader
    data = np.frombuffer(f.read(), dtype=radar_dtype)
ValueError: buffer size must be a multiple of element size

CyanideCN, 2024-03-15:
那应该是其他的格式了，可能又是SA格式的某种变体，无法解决