CyanideCN/PyCINRAD #112: Linux中报错cinrad.error.RadarCalculationError: Cython is not installed

aQua-pl, 2024-05-29:
在WSL中使用conda创建cinrad环境，安装c编译器、cython，并使用pip install cinrad命令在线安装完库后，运行程序报cinrad.error.RadarCalculationError: Cython is not installed错误，在windows系统下同样操作创建环境运行正常，对比检查发现‘~/.conda/envs/cinrad/lib/python3.12/site-packages/cinrad/correct/’目录中缺少_unwrap_2d编译文件，于是手动下载PyCINRAD-1.9.0.zip文件，使用命令pip install ./PyCINRAD-1.9.0.zip重新安装，再次检查‘~/.conda/envs/cinrad/lib/python3.12/site-packages/cinrad/correct/’目录，有新的文件_unwrap_2d.cpython-312-x86_64-linux-gnu.so生成，此时运行程序不再出现ImportError。

CyanideCN, 2024-05-29:
无法复现，linux这边_unwrap_2d只有在装了Cython的情况下才会编译

aQua-pl, 2024-05-29:
我这边WSL使用Ubuntu24.04，按步骤执行的以下命令：
$ conda create -n cinrad-test python=3.12 -y

$ conda activate cinrad-test

$ conda install -y Cython -c conda-forge

$ pip install cinrad

$ ls ~/.conda/envs/cinrad-test/lib/python3.12/site-packages/cinrad/correct/
total 1.0M
drwxr-sr-x 3 aqua aqua 4.0K May 29 14:25 .
drwxr-sr-x 7 aqua aqua 4.0K May 29 14:25 ..
-rw-r--r-- 1 aqua aqua   30 May 29 14:25 __init__.py
drwxr-sr-x 2 aqua aqua 4.0K May 29 14:25 __pycache__
-rw-r--r-- 1 aqua aqua 976K May 29 14:25 _unwrap_2d.c
-rw-r--r-- 1 aqua aqua 1.4K May 29 14:25 dealias.py
-rw-r--r-- 1 aqua aqua  26K May 29 14:25 unwrap_2d_ljmu.c

$ pip install ./PyCINRAD-1.9.0.zip

$ ls ~/.conda/envs/cinrad-test/lib/python3.12/site-packages/cinrad/correct/
total 1.3M
drwxr-sr-x 3 aqua aqua 4.0K May 29 14:29 .
drwxr-sr-x 7 aqua aqua 4.0K May 29 14:29 ..
-rw-r--r-- 1 aqua aqua   29 May 29 14:29 __init__.py
drwxr-sr-x 2 aqua aqua 4.0K May 29 14:29 __pycache__
-rw-r--r-- 1 aqua aqua 978K May 29 14:29 _unwrap_2d.c
-rwxr-xr-x 1 aqua aqua 206K May 29 14:29 _unwrap_2d.cpython-312-x86_64-linux-gnu.so
-rw-r--r-- 1 aqua aqua 1.3K May 29 14:29 dealias.py
-rw-r--r-- 1 aqua aqua  25K May 29 14:29 unwrap_2d_ljmu.c